sdima1357/esp32_usb_soft_host #1: Serial devices (FTDI, CP210X, PL2303, CH340/1 and CDC)

probonopd, 2021-03-10:
@sdima1357 great work :+1: 

Would it be possbile to also support serial devices like

* FTDI FT23... series
* Prolific PL2303
* SiLabs CP210X range
* CH340G / CH341G
* CDC Communications, e.g., Arduino Boards using CDC (UNO, Mega, Leonardo etc.)

similar to https://www.hobbytronics.co.uk/usb-host-serial?

This would be very useful e.g., for 3D printer applications, where the ESP32 could be used to [steer a 3D printer](https://github.com/probonopd/WirelessPrinting) attached over USB.

sdima1357, 2021-03-12:
Unfortunately this program implements only Host side and can work with only Low speed devices (1.5 Mb/s)

probonopd, 2021-03-12:
@sdima1357 that should be plenty for 115,200 bits/s serial which is typical for e.g., 3D printing applications?

sdima1357, 2021-03-12:
It doesn't matter.  USB 1.1 Devices  may be 2 types :  
1:  High speed 12 Mbits/s  - HS device which connect D+ line to 3.3 v for recognize it.
2:  Low speed  1.5 Mbits/s  - LS device which connect D- line to 3.3 v for recognize it.
I can't exchange data with HS device. CH-340,FTDI FT232  - **HS devices**  .They have different speed & different protocol :(

probonopd, 2021-03-12:
Ah, I understand. Thanks for the clarification!

closed 2021-03-12 by sdima1357