sdima1357/esp32_usb_soft_host #2: USB Host MIDI ?

quangvankts, 2021-04-27:
Thank you sdima1357 for the great work.
Is there any chance for a MIDI class driver to be developed ? 
Thank you.

sdima1357, 2021-04-28:
Do You think it can be usable ? Some kind of relay like Wifi <-> midi?  Wifi has unpredictable delays. It think, it will be better use raspberry for these purposes.

quangvankts, 2021-04-28:
I didn't mean wifi <-> MIDI, but sending MIDI commands from app running on the ESP32 over to the USB port. 

Example: right now to do the above, I need a USB Host Shield 2.0 board connect to the ESP32 through SPI. And use this library:
https://github.com/felis/USB_Host_Shield_2.0

The goal sending MIDI over USB in host mode, without any extra hardware. 

Do you think its possible ?

chegewara, 2021-04-28:
Or just use esp32-s2 maybe? USB host support is slowly progressing and i hope it will be available soon.

quangvankts, 2021-04-28:
Yeah esp32-s2 sounds good 
Thanks a lot !

tobozo, 2021-04-28:
Sorry for intruding, I've been asking myself the same question after seeing an old usb midi keyboard being detected.

Good news is the MIDI class driver implemented for the USB Host Shield has some recyclable parts :-)

https://github.com/YuuichiAkagawa/USBH_MIDI 

Possible applications (already seen in the wild): chiptune instrument, music tracker

quangvankts, 2021-05-03:
> Sorry for intruding, I've been asking myself the same question after seeing an old usb midi keyboard being detected.
>
>
>
> Good news is the MIDI class driver implemented for the USB Host Shield has [...]

I'm very interested. Unfortunately writing USB driver is way out of my skill level !

sdima1357, 2021-05-03:
It have checked this options Unfortunately my casio midi keyboard is USB FS  device, but  this code support only USB LS devices :(

tobozo, 2021-05-03:
thanks, I guess it's not worth the effort given the unpredictable delays

closed 2021-05-03 by quangvankts