sdima1357/esp32_usb_soft_host #8: USB Host not working - only getting `Nack`s

sfraint, 2021-09-21:
Latest main doesn't seem to work on my `ESP32`:
```
esp32_usb_soft_host/usb_test$ git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

```
esp32_usb_soft_host/usb_test$ idf.py set-target esp32
...

esp32_usb_soft_host/usb_test$ idf.py flash monitor
...
<several warnings, see below>
...
Hello world end!
b=0x4008ba0c
a=0x3ffafadc
c=0x3ffafb3c
setDelay =0x400d4d1c
0x400d4d1c: setDelay at esp32_usb_soft_host/usb_test/build/../main/usb_host.c:163

pins 16 17 is OK!
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
READ_BOTH_PINS = 0000
cpu freq = 160 MHz
TIME_MULT = 77 
D=   0 4800 bits in 1935.000000 uSec 2.480620 MHz  6 ticks in 2.418750 uS
D= 126 4800 bits in 5651.000000 uSec 0.849407 MHz  6 ticks in 7.063750 uS
D=  63 4800 bits in 3751.000000 uSec 1.279659 MHz  6 ticks in 4.688750 uS
D=  31 4800 bits in 2801.000000 uSec 1.713674 MHz  6 ticks in 3.501250 uS
D=  47 4800 bits in 3271.000000 uSec 1.467441 MHz  6 ticks in 4.088750 uS
D=  39 4800 bits in 3031.000000 uSec 1.583636 MHz  6 ticks in 3.788750 uS
D=  43 4800 bits in 3161.000000 uSec 1.518507 MHz  6 ticks in 3.951250 uS
D=  45 4800 bits in 3210.000000 uSec 1.495327 MHz  6 ticks in 4.012500 uS
D=  44 4800 bits in 3181.000000 uSec 1.508959 MHz  6 ticks in 3.976250 uS
D=  44 4800 bits in 3191.000000 uSec 1.504231 MHz  6 ticks in 3.988750 uS
TRANSMIT_TIME_DELAY = 46 time = 3.988750 error = -0.281250% 
pins -1 -1 is Errors !
pins -1 -1 is Errors !
pins -1 -1 is Errors !
USB0: Ack = 0 Nack = 0 00 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
USB0: Ack = 0 Nack = 44 02 pcurrent->cb_Cmd = 2  state = 3 epCount = 0
USB0: Ack = 0 Nack = 268 02 pcurrent->cb_Cmd = 1  state = 3 epCount = 0
USB0: Ack = 0 Nack = 372 00 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
USB0: Ack = 0 Nack = 372 00 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
USB0: Ack = 0 Nack = 536 02 pcurrent->cb_Cmd = 2  state = 2 epCount = 0
USB0: Ack = 0 Nack = 760 02 pcurrent->cb_Cmd = 2  state = 2 epCount = 0
USB0: Ack = 0 Nack = 764 00 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
```
I tried with a couple different USB devices and none of them worked - kept seeing only `Nack`s.
Any pointers on how to troubleshoot this?

----
I did see a lot of warnings, many about unused var, discarding `volatile`, and `packed` attribute being ignored:
```
[1011/1026] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/usb_test.c.obj
In file included from ../main/usb_test.c:32:
../main/usb_host.h:47:1: warning: 'packed' attribute ignored [-Wattributes]
 } sDevDesc;
 ^
../main/usb_host.h:59:1: warning: 'packed' attribute ignored [-Wattributes]
 } sCfgDesc;
 ^
...
../main/usb_test.c: In function 'kscan0':
../main/usb_test.c:67:10: warning: unused variable 'unum' [-Wunused-variable]
      int unum= msg.src / 4;
          ^~~~
[1016/1026] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/usb_host.c.obj
In file included from ../main/usb_host.c:22:
...
../main/usb_host.c:102:7: warning: initialization discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
      {&GPIO.out_w1tc,&GPIO.out_w1ts},
       ^
../main/usb_host.c:102:22: warning: initialization discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
      {&GPIO.out_w1tc,&GPIO.out_w1ts},
                      ^
...
../main/usb_host.c: In function 'parseImmed':
../main/usb_host.c:342:15: warning: unused variable 'sIntfCount' [-Wunused-variable]
 static int    sIntfCount   = 0;
               ^~~~~~~~~~
../main/usb_host.c:341:15: warning: unused variable 'cfgCount' [-Wunused-variable]
 static int    cfgCount   = 0;
               ^~~~~~~~
...
```
I tried replacing `__packed` with `__attribute__((packed))` which made those warnings go away, but still to no avail.

tobozo, 2021-09-21:
`cpu freq = 160 MHz` looks like your sdkconfig has some leftovers from an ESP32-C3 config

sfraint, 2021-09-21:
funny enough, that seems to be the default speed - any time I run `menuconfig` for a new project, it is sitting on `160` instead of `240`

sdima1357, 2021-09-21:
Did You have problem like this with previous version ?
Unfortunately i don't have  with any of my hardware.  Can You send me more details about Your device  ???
 I checked it with 240,160 & 80 MHz.

sdima1357, 2021-09-21:
**sfraint**
Seems like O2 problem . Thanks for pointing me.

sfraint, 2021-09-21:
Hmm, with corrected `Optimization level`->`O2`, still seeing the same problem. I double-checked the wiring and noticed that `D-` and `D+` were swapped. After correcting the wiring (and replacing all four wires), still not working.

With `D+`->16 and `D-`->17, I see no `Nack`s but also no `Ack`s:
```
USB0: Ack = 0 Nack = 0 01 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
USB0: Ack = 0 Nack = 0 01 pcurrent->cb_Cmd = 0  state = 0 epCount = 0
```
With `D+`->17 and `D-`->16, I see lots of `Nack`s like before:
```
USB0: Ack = 0 Nack = 72 02 pcurrent->cb_Cmd = 2  state = 2 epCount = 0
USB0: Ack = 0 Nack = 296 02 pcurrent->cb_Cmd = 2  state = 2 epCount = 0
USB0: Ack = 0 Nack = 520 02 pcurrent->cb_Cmd = 2  state = 2 epCount = 0
```
Anything I can/should look at without using a logic analyzer?

----

In case it helps, this is my `sdkconfig` compared to `main`:
```
diff --git a/usb_test/sdkconfig b/usb_test/sdkconfig
index 7c6b4ee..a4e3f94 100644
--- a/usb_test/sdkconfig
+++ b/usb_test/sdkconfig
@@ -63,6 +63,7 @@ CONFIG_BOOTLOADER_WDT_TIME_MS=9000
 # CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
 CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
 # CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
+CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
 # end of Bootloader config
 
 #
@@ -83,6 +84,7 @@ CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
 CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
 # CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
 CONFIG_ESPTOOLPY_FLASHMODE="dio"
+# CONFIG_ESPTOOLPY_FLASHFREQ_120M is not set
 # CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
 CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
 # CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
@@ -229,6 +231,12 @@ CONFIG_COAP_LOG_DEFAULT_LEVEL=0
 CONFIG_ADC_DISABLE_DAC=y
 # end of ADC configuration
 
+#
+# MCPWM configuration
+#
+# CONFIG_MCPWM_ISR_IN_IRAM is not set
+# end of MCPWM configuration
+
 #
 # SPI configuration
 #
@@ -278,6 +286,14 @@ CONFIG_EFUSE_CODE_SCHEME_COMPAT_3_4=y
 CONFIG_EFUSE_MAX_BLK_LEN=192
 # end of eFuse Bit Manager
 
+#
+# ESP-NimBLE-CPP configuration
+#
+# CONFIG_NIMBLE_CPP_ENABLE_RETURN_CODE_TEXT is not set
+# CONFIG_NIMBLE_CPP_ENABLE_GAP_EVENT_CODE_TEXT is not set
+# CONFIG_NIMBLE_CPP_ENABLE_ADVERTISMENT_TYPE_TEXT is not set
+# end of ESP-NimBLE-CPP configuration
+
 #
 # ESP-TLS
 #
@@ -298,9 +314,9 @@ CONFIG_ESP32_REV_MIN_0=y
 CONFIG_ESP32_REV_MIN=0
 CONFIG_ESP32_DPORT_WORKAROUND=y
[... 124 lines cut ...]
-#
-# TinyUSB
-#
-# end of TinyUSB
-
 #
 # Unity unit testing library
 #
@@ -1270,6 +1305,7 @@ CONFIG_POST_EVENTS_FROM_IRAM_ISR=y
 CONFIG_FOUR_UNIVERSAL_MAC_ADDRESS=y
 CONFIG_NUMBER_OF_UNIVERSAL_MAC_ADDRESS=4
 CONFIG_ESP_SYSTEM_PD_FLASH=y
+CONFIG_IPC_TASK_STACK_SIZE=1024
 CONFIG_REDUCE_PHY_TX_POWER=y
 # CONFIG_ESP32S2_PANIC_PRINT_HALT is not set
 CONFIG_ESP32S2_PANIC_PRINT_REBOOT=y
@@ -1292,7 +1328,6 @@ CONFIG_TASK_WDT=y
 CONFIG_TASK_WDT_TIMEOUT_S=5
 CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=y
 CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
-CONFIG_IPC_TASK_STACK_SIZE=1024
 CONFIG_TIMER_TASK_STACK_SIZE=3584
 # CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH is not set
 # CONFIG_ESP32_ENABLE_COREDUMP_TO_UART is not set
```

sfraint, 2021-09-21:
And just as a sanity check, my wiring is:
* `Vin` -> `Vbus`
* `TX2` -> `D-`
* `RX2` -> `D+`
* `GND` -> `GND`

sdima1357, 2021-09-21:
Seems that Your device - USB HS device. It can't work with this soft.

With D+->16 and D-->17, I see no Nacks but also no Acks:

USB0: Ack = 0 Nack = 0 **01** pcurrent->cb_Cmd = 0  state = 0 epCount = 0

It put D+ line up on start https://en.wikipedia.org/wiki/USB_(Communications)

closed 2021-09-21 by sdima1357

sfraint, 2021-09-21:
Ahh that was it! Tried another device and it works!