sdima1357/esp32_usb_soft_host #9: Does it work with USB 2.0 keyboards ?

dedesite, 2021-11-05:
Hi,
I'm building an e-ink typewriter based on [Inkplate 6](https://github.com/e-radionicacom/Inkplate-Arduino-library) (just like the [picowriter](https://github.com/deanhuff/picowriter/) project) and just bought a USB 2.0 keyboard for that.
It's a [Motospeed CK62](http://www.motospeed.cc/index.php?ac=article&at=read&did=524) - was the cheapest mecanical 61 keys keyboard I found -  and the manual says that it's a "USB Full Speed 2.0, support up to 1000Hz true reporting rate" whatever that means...
Do you think it could work with your software implementation ?
Thanks,
Andréas

sdima1357, 2021-11-05:
"Does it work with USB 2.0 keyboards"
- If if USB-HS device - definitely no. If it is USB-LS device -You have chance 50/50.

closed 2021-11-05 by sdima1357

dedesite, 2021-11-05:
Thanks for your reply, how do I know if it is a HS or LS USB device ?

dedesite, 2021-11-05:
> But because this keyboard has BT controller You can connect it to ESP32 as BT device.
Yeah I though about that, the problem is that it will need to be charged regularly and will consume a lot of power on the ESP32 board.
I wanted to create a low power device, but it could work for a first prototype I guess.
I don't have a lot of electronic background, do you know if I plug the key board through usb C with an adapter on the GPIO, will it charge through the ESP32 ?

sdima1357, 2021-11-09:
12Mbps is usb hs device. This version of soft can't work with usb hs devices.