sdima1357/esp32_usb_soft_host #10: Does your board provide +5v?

neozeed, 2022-01-19:
its hard to see in the picture but is your board providing a +5v to the USB power pins?  all the esp32 boards I can find don't have a 5v pin to supply output, and I tired using an external 5v power supply but nothing seems to happen at all.  All the keyboards I have tried don't even try to light up.

neozeed, 2022-01-20:
I found the controller you used in spectrum128_esp32_usb_host_pal_tv and it worked using 3.3v.  Im not sure if it's me but it's very difficult to type anything in basic using it.  All the keyboards I have didn't light up or work at all.  I suspect everything has moved off to high speed USB :(

closed 2022-01-20 by neozeed

sdima1357, 2022-01-20:
//its hard to see in the picture
This board(lolin32) has no 5v output pin , only 3.3v 
Some usb devices work unstable with 3.3v power line

neozeed, 2022-01-20:
I just got a new board the LILYGO @ TTGO T7 V1.3 Mini32 V2.0.13 MCU32 WiFi Bluetooth Module Development Board.

it has a pin labeled 5v so I connected that to the USB VCC and not only did the keyboard work (but very poorly, many missing keys and the LEDS never light up) but the wireless one is FAR more responsive.

I tried an external 5v 2a barrel style charger and verified that the 5v gets to the VCC pin but I got nothing from the keyboard.  I’ll have to get a logic analyser I suppose to have any hope of guessing what is going on.

At the same time I have to say that plugging in the AV to a TV and loading files from SD was nothing short of amazing.  Thanks!