sdima1357/esp32_usb_soft_host #12: Is it possible to support USB2.0 fullspeed?

Huckies, 2022-02-25:
Thanks to the excellent work first, it works great with legacy mouse and keyboards which have a 8-Bytes endpoint 0, but it fails when connected to a gaming keyboard with 64-Bytes endpoint 0, no device got enumerated. Is it possible to upgrade the fullspeed support? I guess it will fix the problem.

sdima1357, 2022-02-25:
No, I can't implement full speed. I've tried but without success :( . Too slow port access.

Huckies, 2022-04-02:
Sorry for replying so late. Did you try to set pin drive capability to 3? Even through the GPIO matrix esp32 can run a 20+MHz SPI input or output so I can't figure out why the hardware fails.
I'm not very familiar with the wire-level usb protocol, if you can teach me a little I might be able to help improving this.

sdima1357, 2022-04-02:
"I'm not very familiar with the wire-level usb protocol" 
https://en.wikipedia.org/wiki/USB_(Communications)